alsa-project/alsa-lib #400: [modern hardware] interleaved or plannar?

sylware, 2024-06-08:
To have the less "technical distance" with modern audio hardware (for instance usb audio 2), is it better to have interleaved audio frames each with a pitch, or brutally planar with a pitch per plane?

perexg, 2024-06-10:
The pitch is always similar for one ALSA stream regarless the samples are transferred as interleaved or non-internelaved (planar). Also, USB drivers copies interleaved samples to URB (kernel's USB ring buffer), so there's an extra step in the driver which cannot be avoided. If you use direct ALSA's `hw:CARD,DEV` device, then you have the best transfer method for the given driver/hardware.

closed 2024-06-10 by perexg

sylware, 2024-06-18:
Yeah, I was expecting that.

Since for an audio server, DSP is almost certainly happening, I guess that would be f32/48kHz, plannar multi-channel, with proper SIMD alignment provided by the server upon buffer query.

In means, for a buffer oriented audio server, aka "à la wayland" with something like memfd, and in some ultra-rare scenarios a linux dmabuf.

With the "optional and dynamic" interface discovery, it is possible to have excrutiatingly simple sound server implementations (the "core interfaces" would be something like "optional" availability of the "playback" and/or the "capture" interface).

More complex implementations could provide "an interface for presentation timings" (cf vulkan3D/wayland?), and probably private interfaces for advanced/internal operations (volume, monitoring, etc), private to a desktop implementations.