alsa-project/alsa-lib #404: Soundblaster AE-5 100% Muted on alsamixer every boot!!

DavidRLTG, 2024-08-02:
I am on Linux Mint Wilma (22), 64-bit.
Obviously, I have a Soundblaster AE-5 Sound card
The issue I am encountering is, well, every boot I have to enter Alsamixer and unmute every channel on my Soundblaster AE-5.

alsactl store does nothing, and yes I ran it as superuser.
Also, I'm on pipewire! I think Wilma comes with it preinstalled.

sudo alsactl store is also temporary, the config is lost at every reboot, and doing sudo alsactl restore does nothing.

Any ideas for a fix?

perexg, 2024-08-04:
It looks like a driver issue. You may create a tread on linux-sound@vger.kernel.org list.

closed 2024-08-04 by perexg

DavidRLTG, 2024-08-04:
> It looks like a driver issue. You may create a tread on [linux-sound@vger.kernel.org](mailto:linux-sound@vger.kernel.org) list.

So, do I email them and ask for help, or go to a site?
I apologize, your response has left me confused.