alsa-project/alsa-lib #405: a52 plugin Unknown field card error

DocMAX, 2024-08-04:
```
a52:CARD=Adapter
    USB SPDIF Adapter
    Plugin to convert multichannel stream to A52 (AC3) bitstream

[root@desktop ~]# speaker-test -c6 -Da52

speaker-test 1.2.12

Playback device is a52
Stream parameters are 48000Hz, S16_LE, 6 channels
Using 16 octaves of pink noise
ALSA lib pcm_rate.c:1701:(_snd_pcm_rate_open) Unknown field card
Playback open error: -22,Invalid argument
```
how to fix this error?
running arch linux all updated

perexg, 2024-08-04:
`pcm.a52` is already defined in https://github.com/alsa-project/alsa-plugins/blob/master/a52/60-a52-encoder.conf .

DocMAX, 2024-08-05:
OK, after some trial and error i found this line to be working (removing anything from asound.conf/.asoundrc):
`speaker-test -c6 -D "a52:2,'hw:2,0'"`