alsa-project/alsa-lib #407: Add persistent ID's for soundcards and midi devices (like coreaudio)

sjomae, 2024-08-18:
"This is something where macOS' Coreaudio/MIDI shines. Unlike macOS 
Linux/ALSA has no persistent unique IDs for soundcards or MIDI devices. 
ALSA supports hotplug, and first come first server sequential numeric 
IDs. The best you^Wpipewire can do is keep track of cards by name.

So this is not something pipewire can reliably address, until ALSA get 
support to identify cards by vendor and serial-number, and provide a UUID."

https://lists.linuxaudio.org/hyperkitty/list/linux-audio-dev@lists.linuxaudio.org/thread/OV63UFHRZ2LQYUDADZCUHC2MRA4JWPCU/

perexg, 2024-08-18:
We have 'id' field for this purpose. It contains 15 ASCII characters and can be modified from the user space (e.g. udev rules or user). See `/proc/asound/cards` and the string between `[]`. Also, refer to:

```
cat /sys/class/sound/card0/id
echo "New_name" > /sys/class/sound/card0/id
```

So pipewire should use this string and not the card number for sound card identification.

closed 2024-08-18 by perexg

sjomae, 2024-12-13:
But does this also work for usb midi devices which are not soundcards? Like a launchpad midi controller for example.

perexg, 2024-12-13:
If it's standard USB midi device, then ALSA creates "sound card" for those, too. Just look to `/proc/asound/cards`.

If controller is just USB HID (Human Interface Device), then another kernel driver handles them.