alsa-project/alsa-lib #408: No sound from XonarDX

3bitbrain, 2024-08-21:
No matter what wiki or advice I try, no sound at all from a known good Xonar DX card.
Distro : Gentoo, fresh install.
Always un-muted everything in the console alsamixer.

alsa-lib version 1.2.11

[alsa.txt](https://github.com/user-attachments/files/16697602/alsa.txt)

perexg, 2024-09-06:
Attach output from `alsa-info.sh --no-upload`. Also try to turn on (unmute) spdif or digital output in alsamixer or amixer.

3bitbrain, 2024-09-27:
[alsa-info.txt](https://github.com/user-attachments/files/17170602/alsa-info.txt)

As requested.

Did another wipe the drive and reinstall. Found out kmod is using /etc/modprobe.d/ now. Any conf files in that directory generate errors now.