alsa-project/alsa-lib #410: Some volume controls are swapped

suddenfall, 2024-09-25:
When using amixer and alsamixer, I noticed that the volume controls for certain channels are reversed. I have verified the problem with two sound cards:

Terratec Aureon XFire8.0HD
hifidom DSC8-MAIN

Both are USB sound cards (UAC2).

Both sound cards have the channel map: FL FR FC LFE RL RR SL SR (cat stream0)

The channels FC/LFE are swapped with RL/RR.

I am using Ubuntu 24.04. The version of libasound2t64 is 1.2.11-1build2.

I am happy to help with troubleshooting.

perexg, 2024-10-21:
Could you show me output for `amixer -c 0 contents` (replace zero with the affected card number) when the device is in use? Thanks.

suddenfall, 2024-10-21:
The output is:
```
numid=20,iface=CARD,name='176400Hz Validity'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=21,iface=CARD,name='192000Hz Validity'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=16,iface=CARD,name='44100Hz Validity'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=17,iface=CARD,name='48000Hz Validity'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=18,iface=CARD,name='88200Hz Validity'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=19,iface=CARD,name='96000Hz Validity'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=11,iface=CARD,name='Input Capture Validity'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=15,iface=CARD,name='No Signal Validity'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=10,iface=CARD,name='Onboard Crystal Validity'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=13,iface=MIXER,name='Input Capture Source'
  ; type=ENUMERATED,access=rw------,values=1,items=6
  ; Item #0 'Input 1: card = 1 (SPDIF), port = 1'
  ; Item #1 'Input 2: card = 1 (SPDIF), port = 2'
  ; Item #2 'Input 3: card = 1 (SPDIF), port = 3'
  ; Item #3 'Input 4: card = 1 (SPDIF), port = 4'
  ; Item #4 'Input 5: card = 1 (SPDIF), port = 5'
  ; Item #5 'Input 6: card = 1 (SPDIF), port = 6'
  : values=0
numid=22,iface=MIXER,name='Input Clock Clock Source'
  ; type=ENUMERATED,access=rw------,values=1,items=7
  ; Item #0 'No Signal'
  ; Item #1 '44100Hz'
  ; Item #2 '48000Hz'
  ; Item #3 '88200Hz'
  ; Item #4 '96000Hz'
  ; Item #5 '176400Hz'
  ; Item #6 '192000Hz'
  : values=0
numid=14,iface=MIXER,name='Input Mismatch Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=off
numid=8,iface=MIXER,name='Onboard DAC Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=9,iface=MIXER,name='Onboard Optical Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=12,iface=MIXER,name='Output Clock Clock Source'
  ; type=ENUMERATED,access=rw------,values=1,items=2
  ; Item #0 'Onboard Crystal'
  ; Item #1 'Input Capture'
  : values=0
numid=7,iface=MIXER,name='Output Clock Fallback Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=off
numid=3,iface=MIXER,name='Volume Playback Switch'
  ; type=BOOLEAN,access=rw------,values=8
  : values=on,on,on,on,on,on,on,on
numid=4,iface=MIXER,name='Volume Playback Switch',index=1
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=5,iface=MIXER,name='Volume Playback Volume'
  ; type=INTEGER,access=rw---R--,values=8,min=0,max=127,step=0
  : values=127,127,127,127,127,127,127,127
  | dBminmax-min=-127.00dB,max=0.00dB
numid=6,iface=MIXER,name='Volume Playback Volume',index=1
  ; type=INTEGER,access=rw---R--,values=1,min=0,max=127,step=0
  : values=76
  | dBminmax-min=-127.00dB,max=0.00dB
numid=2,iface=PCM,name='Capture Channel Map'
  ; type=INTEGER,access=r----R--,values=8,min=0,max=36,step=0
  : values=0,0,0,0,0,0,0,0
  | container
    | chmap-fixed=FL,FR,FC,LFE,RL,RR,SL,SR
numid=1,iface=PCM,name='Playback Channel Map'
  ; type=INTEGER,access=r----R--,values=8,min=0,max=36,step=0
  : values=0,0,0,0,0,0,0,0
  | container
    | chmap-fixed=FL,FR,FC,LFE,RL,RR,SL,SR
```

suddenfall, 2024-10-21:
This is the output for the Terratec Aureon XFire8.0HD:
```
numid=12,iface=CARD,name='Clock Source 22 Validity'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=14,iface=CARD,name='Clock Source 23 Validity'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=23,iface=CARD,name='Clock Source 24 Validity'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=24,iface=CARD,name='Clock Source 25 Validity'
  ; type=BOOLEAN,access=r-------,values=1
  : values=off
numid=13,iface=MIXER,name='PCM Playback Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=22,iface=MIXER,name='PCM Capture Source'
  ; type=ENUMERATED,access=rw------,values=1,items=3
  ; Item #0 'Mic'
  ; Item #1 'Line'
  ; Item #2 'Mixer'
  : values=0
numid=20,iface=MIXER,name='PCM Capture Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=21,iface=MIXER,name='PCM Capture Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=28,step=0
  : values=16,16
  | dBminmax-min=-16.00dB,max=12.00dB
numid=8,iface=MIXER,name='Line Playback Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=off
numid=9,iface=MIXER,name='Line Playback Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=31,step=0
  : values=23,23
  | dBminmax-min=-34.50dB,max=12.00dB
numid=18,iface=MIXER,name='Line Capture Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=off
numid=19,iface=MIXER,name='Line Capture Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=28,step=0
  : values=0,0
  | dBminmax-min=-16.00dB,max=12.00dB
numid=6,iface=MIXER,name='Mic Playback Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=off
numid=7,iface=MIXER,name='Mic Playback Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=31,step=0
  : values=23,23
  | dBminmax-min=-34.50dB,max=12.00dB
numid=15,iface=MIXER,name='Mic Capture Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=16,iface=MIXER,name='Mic Capture Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=28,step=0
  : values=28,28
  | dBminmax-min=-16.00dB,max=12.00dB
numid=17,iface=MIXER,name='Auto Gain Control'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=10,iface=MIXER,name='Speaker Playback Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=11,iface=MIXER,name='Speaker Playback Volume'
  ; type=INTEGER,access=rw---R--,values=8,min=0,max=62,step=0
  : values=39,39,39,39,39,39,39,39
  | dBminmax-min=-62.00dB,max=0.00dB
numid=4,iface=PCM,name='Capture Channel Map'
  ; type=INTEGER,access=r----R--,values=2,min=0,max=36,step=0
  : values=0,0
  | container
    | chmap-fixed=FL,FR
numid=1,iface=PCM,name='Playback Channel Map'
  ; type=INTEGER,access=r----R--,values=8,min=0,max=36,step=0
  : values=0,0,0,0,0,0,0,0
  | container
    | chmap-fixed=FL,FR,FC,LFE,RL,RR,SL,SR
    | chmap-fixed=FL,FR
    | chmap-fixed=FL,FR,FC,LFE,RL,RR
    | chmap-fixed=FL,FR,RL,RR
numid=5,iface=PCM,name='Capture Channel Map',device=1
  ; type=INTEGER,access=r----R--,values=2,min=0,max=36,step=0
  : values=0,0
  | container
    | chmap-fixed=FL,FR
numid=2,iface=PCM,name='Playback Channel Map',device=1
  ; type=INTEGER,access=r----R--,values=2,min=0,max=36,step=0
  : values=0,0
  | container
    | chmap-fixed=FL,FR
numid=3,iface=PCM,name='Playback Channel Map',device=2
  ; type=INTEGER,access=r----R--,values=2,min=0,max=36,step=0
  : values=0,0
  | container
    | chmap-fixed=FL,FR

```

atazx, 2025-02-21:
same problem with Asus Xonar SE 
`0b05:189d ASUSTek Computer, Inc. Xonar SoundCard
`

`amixer -c 3 contents
numid=10,iface=CARD,name='Clock Source 21 Validity'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=15,iface=CARD,name='Clock Source 22 Validity'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=18,iface=CARD,name='Clock Source 23 Validity'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=30,iface=CARD,name='Clock Source 24 Validity'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=35,iface=CARD,name='Clock Source 25 Validity'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=16,iface=CARD,name='Headphone - Output Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=off
numid=19,iface=CARD,name='IEC958 In - Output Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=24,iface=CARD,name='Line - Input Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=20,iface=CARD,name='Mic - Input Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=31,iface=CARD,name='Mic - Input Jack',index=1
  ; type=BOOLEAN,access=r-------,values=1
  : values=off
numid=11,iface=CARD,name='Speaker - Output Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
numid=7,iface=MIXER,name='PCM Playback Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=12,iface=MIXER,name='PCM Playback Switch',index=1
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=17,iface=MIXER,name='PCM Playback Switch',index=2
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=8,iface=MIXER,name='PCM Playback Volume'
  ; type=INTEGER,access=rw---R--,values=6,min=0,max=87,step=0
  : values=87,87,87,87,85,85
  | dBminmax-min=-65.25dB,max=0.00dB
numid=13,iface=MIXER,name='PCM Playback Volume',index=1
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=87,step=0
  : values=59,59
  | dBminmax-min=-65.25dB,max=0.00dB
numid=29,iface=MIXER,name='PCM Capture Source'
  ; type=ENUMERATED,access=rw------,values=1,items=3
  ; Item #0 'Mic'
  ; Item #1 'Line'
  ; Item #2 'Mixer'
  : values=0
numid=27,iface=MIXER,name='PCM Capture Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=28,iface=MIXER,name='PCM Capture Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=63,step=0
  : values=23,23
  | dBminmax-min=-17.25dB,max=30.00dB
numid=25,iface=MIXER,name='Line Capture Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=off
numid=26,iface=MIXER,name='Line Capture Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=23,step=0
  : values=0,0
  | dBminmax-min=-17.25dB,max=0.00dB
numid=21,iface=MIXER,name='Mic Capture Switch'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=32,iface=MIXER,name='Mic Capture Switch',index=1
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=22,iface=MIXER,name='Mic Capture Volume'
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=63,step=0
  : values=63,63
  | dBminmax-min=-17.25dB,max=30.00dB
numid=33,iface=MIXER,name='Mic Capture Volume',index=1
  ; type=INTEGER,access=rw---R--,values=2,min=0,max=63,step=0
  : values=39,39
  | dBminmax-min=-17.25dB,max=30.00dB
numid=23,iface=MIXER,name='Auto Gain Control'
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=34,iface=MIXER,name='Auto Gain Control',index=1
  ; type=BOOLEAN,access=rw------,values=1
  : values=on
numid=9,iface=MIXER,name='Input Gain Pad Control'
  ; type=INTEGER,access=rw------,values=1,min=0,max=2,step=0
  : values=1
numid=14,iface=MIXER,name='Input Gain Pad Control',index=1
  ; type=INTEGER,access=rw------,values=2,min=0,max=1,step=0
  : values=0,0
numid=5,iface=PCM,name='Capture Channel Map'
  ; type=INTEGER,access=r--v-R--,values=2,min=0,max=36,step=0
  : values=3,4
  | container
    | chmap-fixed=FL,FR
numid=1,iface=PCM,name='Playback Channel Map'
  ; type=INTEGER,access=r--v-R--,values=6,min=0,max=36,step=0
  : values=3,4,7,8,9,10
  | container
    | chmap-fixed=FL,FR,FC,LFE,SL,SR
    | chmap-fixed=FL,FR
    | chmap-fixed=FL,FR,RL,RR
numid=6,iface=PCM,name='Capture Channel Map',device=1
  ; type=INTEGER,access=r--v-R--,values=2,min=0,max=36,step=0
  : values=0,0
  | container
    | chmap-fixed=FL,FR
numid=2,iface=PCM,name='Playback Channel Map',device=1
  ; type=INTEGER,access=r--v-R--,values=2,min=0,max=36,step=0
  : values=0,0
  | container
    | chmap-fixed=FL,FR
numid=3,iface=PCM,name='Playback Channel Map',device=2
  ; type=INTEGER,access=r--v-R--,values=2,min=0,max=36,step=0
  : values=0,0
  | container
    | chmap-fixed=FL,FR
numid=4,iface=PCM,name='Playback Channel Map',device=3
  ; type=INTEGER,access=r--v-R--,values=2,min=0,max=36,step=0
  : values=0,0
  | container
    | chmap-fixed=FL,FR
`