alsa-project/alsa-lib #411: MUSL static build - conf.c still insists on using dynamic link libraries

pietervandermeer, 2024-10-16:
alsa-lib v 1.2.12

configured for musl static build like so:

```CC=arm-linux-musleabihf-gcc ./configure --enable-shared=no --enable-static=yes --with-pic --host=armv7-unknown-linux-musleabihf --prefix=/home/vagrant/alsa-lib-1.2.12/build --libdir=/usr/lib/arm-linux-musleabihf --includedir=/usr/include/arm-linux-musleabihf/```

This builds and links fine against my executable. But when I run it, there's an issue:

```
INFO opening alsa audio playback device [hw:CARD=sndrpihifiberry,DEV=0]
ALSA lib conf.c:3558:(snd_config_hooks_call) Cannot open shared library (null) ((null): Dynamic loading not supported)
ALSA lib conf.c:4026:(snd_config_update_r) hooks failed, removing configuration
WARN Not found playback device, error ALSA function 'snd_pcm_open' failed with error 'No such file or directory (2)'. Trying next..
```

Note the "cannot open shared library". Well, yes, because you should be configured as 100% static, as per configure arguments ```--enable-shared=no --enable-static=yes```

Upon inspecting the code in ```conf.c```, there are various functions like ```snd_config_hooks_call()``` that just call ```snd_dlopen()``` and ```snd_dlsym()```.

Right now I'm just replacing these with ```strcmp``` on the function_name followed by a static function call to the respective function. But I don't know how much work I'll need, and of course it is really hacky. 

It would be great if a 100% static build was supported.

pietervandermeer, 2024-10-17:
Ah, found the issue. ```./configure``` needs to be done without the ```--with-pic``. Apparently it's not compatible with ```--enable-static=yes```... Strange workaround to make a static build, though.. The static build kinda fakes things to get ```dlopen``` ```dlsym```working. 

Oh well, it's free software. I'm not complaining.. Although ```./configure --help``` could definitely mention this.

perexg, 2024-10-21:
With the above patch:

```
$ ./configure --enable-shared=no --enable-static=yes --with-pic
...
checking for library version... major 1 minor 2 subminor 13 extrastr pre extraver 1
  Please, do not enable PIC (--with-pic) for static library. Consider
  to remove this option or use PIE flags.
```

closed 2024-10-21 by perexg