alsa-project/alsa-lib #412: snd_pcm_hw_open open '/dev/snd/pcmC0D0p' failed (-16)

SVENsuzhou, 2024-10-21:
Hi, I am using pjsip to make a sip call function. But I got the following error in alsa-lib.
```
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.315           pjsua_call.c !Answering call 0: code=200
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.315      inv0x7f989c003fc8  ..SDP negotiation done: Success
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.316          pjsua_media.c  ...Call 0: updating media..
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.316          pjsua_media.c  .....Media stream call00:0 is destroyed
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.316      udp0x7f989c010710  ....UDP media transport started
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.316            pjsua_aud.c  ....Audio channel update..
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.316      udp0x7f989c010710  .....UDP media transport attached
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.317     strm0x5559ce2eb538  .....Encoder stream started
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.317     strm0x5559ce2eb538  .....Decoder stream started
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.317          pjsua_media.c  ....Audio updated, stream #0: PCMA (sendrecv)
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.317            pjsua_aud.c  ...Conf connect: 1 --> 0
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.317            pjsua_aud.c  ....Set sound device: capture=-1, playback=-2, mode=0, use_default_settings=0
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.317            pjsua_aud.c  .....Opening sound device (speaker + mic) PCM@16000/1/20ms
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.318             alsa_dev.c  ......ALSA lib /usr/src/debug/alsa-lib/1.2.11/src/pcm/pcm_hw.c:1785:(snd_pcm_hw_open) open '/dev/snd/pcmC0D0p' failed (-16): Device o
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.318            pjsua_aud.c  .....Opening sound device (speaker + mic) PCM@44100/1/20ms
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: start media transmit failed: pjsua_conf_connect(id, sink.id) error: Unknown error from audio driver (PJMEDIA_EAUD_SYSERR) (status=420002) [/usr/src/debug/pjproject/2.14.1/pjsip/src/pjsua2/media.cpp:218]
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.318             alsa_dev.c  ......ALSA lib /usr/src/debug/alsa-lib/1.2.11/src/pcm/pcm_hw.c:1785:(snd_pcm_hw_open) open '/dev/snd/pcmC0D0p' failed (-16): Device o
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.318            pjsua_aud.c  .....Opening sound device (speaker + mic) PCM@48000/1/20ms
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.319             alsa_dev.c  ......ALSA lib /usr/src/debug/alsa-lib/1.2.11/src/pcm/pcm_hw.c:1785:(snd_pcm_hw_open) open '/dev/snd/pcmC0D0p' failed (-16): Device o
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.319            pjsua_aud.c  .....Opening sound device (speaker + mic) PCM@32000/1/20ms
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.319             alsa_dev.c  ......ALSA lib /usr/src/debug/alsa-lib/1.2.11/src/pcm/pcm_hw.c:1785:(snd_pcm_hw_open) open '/dev/snd/pcmC0D0p' failed (-16): Device o
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.319            pjsua_aud.c  .....Opening sound device (speaker + mic) PCM@16000/1/20ms
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.319             alsa_dev.c  ......ALSA lib /usr/src/debug/alsa-lib/1.2.11/src/pcm/pcm_hw.c:1785:(snd_pcm_hw_open) open '/dev/snd/pcmC0D0p' failed (-16): Device o
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.319            pjsua_aud.c  .....Opening sound device (speaker + mic) PCM@8000/1/20ms
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.320             alsa_dev.c  ......ALSA lib /usr/src/debug/alsa-lib/1.2.11/src/pcm/pcm_hw.c:1785:(snd_pcm_hw_open) open '/dev/snd/pcmC0D0p' failed (-16): Device o
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.320            pjsua_aud.c  .....Unable to open sound device: Unknown error from audio driver (PJMEDIA_EAUD_SYSERR) [status=420002]
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.320            pjsua_aud.c  ....Error opening sound device: Unknown error from audio driver (PJMEDIA_EAUD_SYSERR) [status=420002]
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.320              media.cpp  ...pjsua_conf_connect(id, sink.id) error: Unknown error from audio driver (PJMEDIA_EAUD_SYSERR) (status=420002) [/usr/src/debug/pjproject/2.14.1/pjsip/src/pjsua2/media.cpp:218]
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: 08:14:36.321           pjsua_core.c  ....TX 966 bytes Response msg 200/INVITE/cseq=30946 (tdta0x5559ce2e7378) to UDP 10.57.0.200:5060:
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: SIP/2.0 200 OK
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: Via: SIP/2.0/UDP 10.57.0.200:5060;rport=5060;received=10.57.0.200;branch=z9hG4bKPje5a12777-db72-4db0-a5a5-a72743589a27
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: Call-ID: 0bc91889-099c-42de-beb2-4fedefac6e7f
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: answer incoming call, remote uri: <sip:passenger_pei@10.57.0.200>, state: CONNECTING
Oct 21 08:14:36 tos-hmi-02 ads_uartboardd[1404]: From: <sip:passenger_pei@10.57.0.200>;tag=9efc50b5-c57a-4aa7-823e-2c3df064a27c
```
I try to use `lsof /dev/snd/*` to check which process hold the file handle. And I got nothing. Can you give me some leads on troubleshooting?

perexg, 2024-10-21:
Error -16 means -EBUSY. The error is from the driver not alsa-lib. Does `aplay -D plughw:0 -f dat /dev/zero` command work on your platform ?

SVENsuzhou, 2024-10-21:
Yes, it works.
```
root@tos-hmi-02:~# aplay -D plughw:0 -f dat /dev/zero
Playing raw data '/dev/zero' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo

```

perexg, 2024-10-21:
It would be probably best to analyze which parameters are used with this particular application. The `snd_pcm_dump` function may be helpful.

SVENsuzhou, 2024-10-22:
Thanks for your help. I think I have catched the mistake. There is indeed a file handle contention problem, it's just short enough that the lsof command can't detect it.

closed 2024-10-22 by SVENsuzhou