alsa-project/alsa-lib #413: ALSA error: snd_pcm_hw_params_set_format failed: Invalid argument.

linuxandos2, 2024-10-21:
Playing a flac with a samplerate higer than 96000 with audacious gives: 

> _ALSA error: snd_pcm_hw_params_set_format failed: Invalid argument._

Output device for audacious is hw:1.0 - USB Audio
the device is a soudbaster blasterx G6.

output of _cat /proc/asound/card1/stream0_

> Creative Technology Ltd Sound BlasterX G6 at usb-0000:00:14.0-4.1, high speed :
> USB Audio
> 
> Playback:
>   Status: Stop
>   Interface 1
>     Altset 1
>     Format: S24_3LE
>     Channels: 2
>     Endpoint: 0x01 (1 OUT) (ASYNC)
>     Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
>     Data packet interval: 125 us
>     Bits: 24
>     Channel map: FL FR
>     Sync Endpoint: 0x81 (1 IN)
>     Sync EP Interface: 1
>     Sync EP Altset: 1
>     Implicit Feedback Mode: No
>   Interface 1
>     Altset 2
>     Format: S32_LE
>     Channels: 2
>     Endpoint: 0x01 (1 OUT) (ASYNC)
>     Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
>     Data packet interval: 125 us
>     Bits: 32
>     Channel map: FL FR
>     Sync Endpoint: 0x81 (1 IN)
>     Sync EP Interface: 1
>     Sync EP Altset: 2
>     Implicit Feedback Mode: No
> 
> Capture:
>   Status: Stop
>   Interface 2
>     Altset 1
>     Format: S24_3LE
>     Channels: 2
>     Endpoint: 0x82 (2 IN) (ASYNC)
>     Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
>     Data packet interval: 125 us
>     Bits: 24
>     Channel map: FL FR
>   Interface 2
>     Altset 2
>     Format: S32_LE
>     Channels: 2
>     Endpoint: 0x82 (2 IN) (ASYNC)
>     Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
>     Data packet interval: 125 us
>     Bits: 32
>     Channel map: FL FR
> 
linux: gentoo
kernel: 6.9.9-gentoo-x86_64
alsa-lib-1.2.12
alsa-topology-conf-1.2.5.1
alsa-ucm-conf-1.2.12
alsa-plugins-1.2.12
alsa-utils-1.2.12

selecting  

> pulse - PulseAudio Sound Server

 in audacious results in no sound and audacious hangs.
Pipewire and wireplumber are used.

On an old gentoo- system on another partition all samplerates play successfully.

kernel: linux-6.8.1-gentoo
the versions of alsa are:
alsa-lib-1.2.10-r2
alsa-topology-conf-1.2.5.1
alsa-ucm-conf-1.2.10-r1
alsa-plugins-1.2.7.1-r1
alsa-utils-1.2.10-r1

On the new system the 'old' kernel linux-6.8.1-gentoo is also available.
Rolled back alsa-lib from alsa-lib-1.2.12 to alsa-lib-1.2.10 and booted kernel kernel linux-6.8.1-gentoo
Still same problem.

perexg, 2024-10-21:
It looks like a kernel driver issue (but maybe not even the sound driver). Try to connect your USB audio device to another USB port (like 2.0 only) on your system. Some USB host drivers have issues with the bandwidth calculation. Also try to reproduce your issue with the `aplay` or some basic ALSA utility. Like `aplay -D hw:0 -c 1 -f S24_LE -r 96000 -c 2 /dev/zero` . Replace zero in `hw:0` with your card number.

tiwai, 2024-10-22:
Make sure that no input stream is running on the same device while you test different parameters for the output.
Since the device runs in the implicit feedback mode, the parameters are tied for both input and output.

linuxandos2, 2024-10-22:
Yes there is no input stream running, dureing both tests.
cat /proc/asound/card1/stream0:

> Capture:
  Status: Stop
  Interface 2
    Altset 1
    Format: S24_3LE
    Channels: 2
    Endpoint: 0x82 (2 IN) (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
    Bits: 24
    Channel map: FL FR
  Interface 2
    Altset 2
    Format: S32_LE
    Channels: 2
    Endpoint: 0x82 (2 IN) (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
    Bits: 32
    Channel map: FL FR

linuxandos2, 2024-10-22:
With buffer-size=1000000 it works.
Where can I change the buffersize globaly?

linuxandos2, 2024-10-22:
Test with plughw, rate 96000, buffer-size=1000000 ==>OK
aplay -D plughw:1,0 -c 1 -f S24_3LE -r 96000 -c 2 --buffer-size=1000000 /dev/zero
> Playing raw data '/dev/zero' : Signed 24 bit Little Endian in 3bytes, Rate 96000 Hz, Stereo

Test with plughw, rate 96000, buffer-size=1000000 **==>NOK**
aplay -D plughw:1,0 -c 1 -f S24_3LE -r 192000 -c 2 --buffer-size=1000000 /dev/zero
> Playing raw data '/dev/zero' : Signed 24 bit Little Endian in 3bytes, Rate 192000 Hz, Stereo
aplay: set_params:1456: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S24_3LE
SUBFORMAT:  STD
SAMPLE_BITS: 24
FRAME_BITS: 48
CHANNELS: 2
RATE: 192000
PERIOD_TIME: 1000000
PERIOD_SIZE: 192000
PERIOD_BYTES: 1152000
PERIODS: 2
BUFFER_TIME: 2000000
BUFFER_SIZE: 384000
BUFFER_BYTES: 2304000
TICK_TIME: 0

linuxandos2, 2024-10-22:
Also tryed different period sizes.

linuxandos2, 2024-10-24:
Finally installed a fresh gentoo system on a different partition:
Problem does not occur any more.
If you want to further investigate let my know.
Up to you to close.
And thanks alot.

closed 2024-11-12 by perexg