alsa-project/alsa-lib #414: AlSA problem , audio on/off across multiple ubutnu

Kermilli, 2024-10-22:
Hi, 

I've been having problems with audio on ubuntu on and off. I'm not a professional but everything seems to point at alsa . The pulse troubleshooting page at some point recommends to run alsa directly/bypass pulse with the command `mplayer -ao alsa:device=hw=0 yourchosentrack.mp3`  which  gives 
`[AO_ALSA] alsa-lib: pcm_hw.c:1715:(snd_pcm_hw_open) open '/dev/snd/pcmC0D0p' failed (-2): No such file or directory
[AO_ALSA] Playback open error: No such file or directory
Failed to initialize audio driver 'alsa:device=hw=0'
Could not open/initialize audio device -> no sound.
Audio: no sound
Video: no video
` 
sudo pulseaudio -vvv gives

`E: [pulseaudio] module-alsa-card.c: Failed to find a working profile.
E: [pulseaudio] module.c: Failed to load module "module-alsa-card" (argument: "device_id="1" name="pci-0000_00_1b.0" card_name="alsa_card.pci-0000_00_1b.0" namereg_fail=false tsched=yes fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1""): initialization failed.`

 
I've tried everything . The only time it worked was when I added options snd-hda-intel model=auto to  /etc/modprobe.d/alsa-base.conf. , but even then it stopped working after some time when I was listening to audio.

Any help would be greatly appreciated!

perexg, 2024-10-22:
It looks like a kernel driver issue. Attach `dmesg`, but it will be probably better to continue on `linux-sound@vger.kernel.org` mailing list.

Kermilli, 2024-10-22:
`l-Precision-M6800:~$ dmesg
[    0.000000] microcode: microcode updated early to revision 0x28, date = 2019-11-12
[    0.000000] Linux version 5.15.0-124-generic (buildd@lcy02-amd64-092) (gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #134~20.04.1-Ubuntu SMP Tue Oct 1 15:27:33 UTC 2024 (Ubuntu 5.15.0-124.134~20.04.1-generic 5.15.163)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.15.0-124-generic root=UUID=ad9a2e46-306c-452c-b97c-2e33b32213bf ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000004efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000004f000-0x000000000004ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000050000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000bf0b7fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bf0b8000-0x00000000bf0befff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000bf0bf000-0x00000000bfa93fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000bfa94000-0x00000000bff72fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000bff73000-0x00000000c7db2fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000c7db3000-0x00000000c7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000c8000000-0x00000000c8756fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000c8757000-0x00000000c87fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000c8800000-0x00000000c8f92fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000c8f93000-0x00000000c8ffffff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000c9000000-0x00000000ca6f4fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ca6f5000-0x00000000ca7fffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000ca800000-0x00000000cbcdbfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000cbcdc000-0x00000000cbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000cd000000-0x00000000cf1fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000082edfffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0xc5329018-0xc5352457] usable ==> usable
[    0.000000] e820: update [mem 0xc5329018-0xc5352457] usable ==> usable
[    0.000000] e820: update [mem 0xc5318018-0xc5328057] usable ==> usable
[    0.000000] e820: update [mem 0xc5318018-0xc5328057] usable ==> usable
[    0.000000] e820: update [mem 0xc5307018-0xc5317457] usable ==> usable
[    0.000000] e820: update [mem 0xc5307018-0xc5317457] usable ==> usable
[    0.000000] e820: update [mem 0xc52e9018-0xc5306c57] usable ==> usable
[    0.000000] e820: update [mem 0xc52e9018-0xc5306c57] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000004efff] usable
[    0.000000] reserve setup_data: [mem 0x000000000004f000-0x000000000004ffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000050000-0x000000000009efff] usable
[    0.000000] reserve setup_data: [mem 0x000000000009f000-0x000000000009ffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x00000000bf0b7fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000bf0b8000-0x00000000bf0befff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x00000000bf0bf000-0x00000000bfa93fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000bfa94000-0x00000000bff72fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000bff73000-0x00000000c52e9017] usable
[    0.000000] reserve setup_data: [mem 0x00000000c52e9018-0x00000000c5306c57] usable
[    0.000000] reserve setup_data: [mem 0x00000000c5306c58-0x00000000c5307017] usable
[    0.000000] reserve setup_data: [mem 0x00000000c5307018-0x00000000c5317457] usable
[    0.000000] reserve setup_data: [mem 0x00000000c5317458-0x00000000c5318017] usable
[    0.000000] reserve setup_data: [mem 0x00000000c5318018-0x00000000c5328057] usable
[    0.000000] reserve setup_data: [mem 0x00000000c5328058-0x00000000c5329017] usable
[    0.000000] reserve setup_data: [mem 0x00000000c5329018-0x00000000c5352457] usable
[    0.000000] reserve setup_data: [mem 0x00000000c5352458-0x00000000c7db2fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000c7db3000-0x00000000c7ffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000c8000000-0x00000000c8756fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000c8757000-0x00000000c87fffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000c8800000-0x00000000c8f92fff] usable
[    0.000000] reserve setup_data: [mem 0x00000000c8f93000-0x00000000c8ffffff] ACPI data
[... cut ...]
[   25.312405] audit: type=1400 audit(1729614220.440:71): apparmor="DENIED" operation="open" profile="snap.snap-store.ubuntu-software" name="/etc/appstream.conf" pid=2159 comm="snap-store" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
`
There is an error on `snd_hda_intel 0000:00:1b.0: no codecs found! `

If it helps  there is no HDA Intel PCH under aplay -l and missing files proc/asound/ for the appropriate card. Also before I could change the volume of the card on alsamixer but now it shows `This sound device does not have any controls. `

Also I can play sudo mplayer -ao alsa:device=hdmi ~/Music/sample.mp3 with the HDMI card of HDA Intel HDMI

Kermilli, 2024-10-22:
I'm sorry , what should I do further ?

Kermilli, 2024-10-22:
I fixed my issue by trying to copy my respective PCH files from a live cd and when I booted up on my SSD audio was available again. Hope it helps anyone.

If it helps further anyone further , where should I report this to @perexg ?

Kermilli, 2024-10-22:
I tried copying my respective PCH files from a live cd and when I booted up on my SSD audio was available again. Hope it helps anyone.

borine, 2024-10-23:
> where should I report this

If some ALSA files are missing from an Ubuntu installation, then I would suggest that this should be reported as an issue to Ubuntu. (I use Ubuntu, and have never seen such a problem; is it possible that these files, you do not say which ones, were accidentally deleted after the original installation?)

Kermilli, 2024-10-23:
I don't completely understand how asla work. All I can see is that the card that runs my internal speakers doesn't show up on proc/asound .As far as I can tell proc/asound is a virtual system and nothing can be copied into it . Up until now I could load up a live cd and Ubuntu would recognize my card on startup , but even now the sound doesn't work on live cd. 

What is going on ? As far as I know proc/asound is related to alsa .

Kermilli, 2024-10-23:
@borine  Oh man I think my CMOS battery is very low and is causing problem with my audio drivers loading up . I tried hard resetting/holding down the power button without the onboard battery for two minutes and on startup I got audio again. 

I had previously tried but held it only for 1 minute as recommended by the web and it didn't work.  I had only once a problem with my time and bios configuration reseting and It didn't occur to me that it might be a CMOS battery problems.

Thank you very much for your responses!

closed 2024-11-07 by perexg