alsa-project/alsa-lib #424: Linux Mint Cinnamon 22 Soundblaster AE-9 no sound

Starfoxfs, 2024-11-30:
Hi,

i have fresh installed Linux Mint Cinnamon 22 Kernel 6.8.0.49-generic

I can´t get the Soundblaster AE-9 to work. Creative Labs has no Drivers for Linux and points to the alsa community.

Is there any change to get support for the Soundblaster AE-9 in the future ?

I think the big Problem is the external DAC for Headphones from the Soundblaster AE-9. In Windows this "clicks" on startup and activates the DAC, but in Linux this is all dead.

Starfoxfs, 2024-12-02:
Here are some additional Information:

`inxi -Fxxxrz
System:
  Kernel: 6.8.0-49-generic arch: x86_64 bits: 64 compiler: gcc v: 13.2.0
    clocksource: tsc
  Desktop: Cinnamon v: 6.2.9 tk: GTK v: 3.24.41 wm: Muffin v: 6.2.0 vt: 7
    dm: LightDM v: 1.30.0 Distro: Linux Mint 22 Wilma base: Ubuntu 24.04 noble
Machine:
  Type: Desktop System: ASUS product: N/A v: N/A serial: <superuser required>
  Mobo: ASUSTeK model: TUF H370-PRO GAMING v: Rev 1.xx
    serial: <superuser required> part-nu: ASUS_MB_CNL
    UEFI-[Legacy]: American Megatrends v: 3101 date: 09/10/2021
CPU:
  Info: 6-core model: Intel Core i5-9400F bits: 64 type: MCP
    smt: <unsupported> arch: Coffee Lake rev: A cache: L1: 384 KiB L2: 1.5 MiB
    L3: 9 MiB
  Speed (MHz): avg: 800 min/max: 800/4100 cores: 1: 800 2: 800 3: 800 4: 800
    5: 800 6: 800 bogomips: 34798
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: NVIDIA GP104 [GeForce GTX 1070] vendor: CardExpert driver: nvidia
    v: 550.120 arch: Pascal pcie: speed: 2.5 GT/s lanes: 16 ports: active: none
    off: DVI-D-1 empty: DP-1, DP-2, DP-3, HDMI-A-1 bus-ID: 01:00.0
    chip-ID: 10de:1b81 class-ID: 0300
  Display: x11 server: X.Org v: 21.1.11 with: Xwayland v: 23.2.6 driver: X:
    loaded: nvidia unloaded: fbdev,modesetting,nouveau,vesa
    gpu: nvidia,nvidia-nvswitch display-ID: :0 screens: 1
  Screen-1: 0 s-res: 1920x1200 s-dpi: 84 s-size: 581x363mm (22.87x14.29")
    s-diag: 685mm (26.97")
  Monitor-1: DVI-D-0 res: 1920x1200 hz: 60 dpi: 84
    size: 582x364mm (22.91x14.33") diag: 686mm (27.03") modes: N/A
  API: EGL v: 1.5 hw: drv: nvidia platforms: device: 0 drv: nvidia device: 2
    drv: swrast gbm: drv: nvidia surfaceless: drv: nvidia x11: drv: nvidia
    inactive: wayland,device-1
  API: OpenGL v: 4.6.0 compat-v: 4.5 vendor: nvidia mesa v: 550.120
    glx-v: 1.4 direct-render: yes renderer: NVIDIA GeForce GTX 1070/PCIe/SSE2
Audio:
  Device-1: NVIDIA GP104 High Definition Audio vendor: CardExpert
    driver: snd_hda_intel v: kernel pcie: speed: 8 GT/s lanes: 16
    bus-ID: 01:00.1 chip-ID: 10de:10f0 class-ID: 0403
  Device-2: Creative Labs driver: snd_hda_intel v: kernel bus-ID: 06:00.0
    chip-ID: 1102:0010 class-ID: 0403
  API: ALSA v: k6.8.0-49-generic status: kernel-api
  Server-1: PipeWire v: 1.0.5 status: active with: 1: pipewire-pulse
    status: active 2: wireplumber status: active
  Server-2: PulseAudio v: 16.1 status: off (using pipewire-pulse)
Network:
  Device-1: Intel Ethernet I219-V vendor: ASUSTeK driver: e1000e v: kernel
    port: N/A bus-ID: 00:1f.6 chip-ID: 8086:15bc class-ID: 0200
  IF: eno1 state: up speed: 1000 Mbps duplex: full mac: <filter>
Drives:
  Local Storage: total: 2.96 TiB used: 1019.51 GiB (33.7%)
  ID-1: /dev/nvme0n1 vendor: Crucial model: CT500P3PSSD8 size: 465.76 GiB
    speed: 63.2 Gb/s lanes: 4 tech: SSD serial: <filter> fw-rev: P9CR413
    temp: 36.9 C scheme: GPT
  ID-2: /dev/sda vendor: Crucial model: CT500MX500SSD1 size: 465.76 GiB
    speed: 6.0 Gb/s tech: SSD serial: <filter> fw-rev: 023 scheme: MBR
  ID-3: /dev/sdb vendor: Western Digital model: WD2005FBYZ-01YCBB3
    size: 1.82 TiB speed: 6.0 Gb/s tech: HDD rpm: 7200 serial: <filter>
    fw-rev: RR09 scheme: GPT
  ID-4: /dev/sdc vendor: Western Digital model: WD Blue SA510 2.5 250GB
    size: 232.89 GiB speed: 6.0 Gb/s tech: SSD serial: <filter> fw-rev: 6100
    scheme: GPT
Partition:
  ID-1: / size: 91.33 GiB used: 17.9 GiB (19.6%) fs: ext4 dev: /dev/sdc2
  ID-2: /boot/efi size: 96 MiB used: 36.7 MiB (38.2%) fs: vfat
    dev: /dev/nvme0n1p1
  ID-3: /home size: 128.4 GiB used: 11.09 GiB (8.6%) fs: ext4 dev: /dev/sdc4
Swap:
[... 46 lines cut ...]
00:00.0 Host bridge: Intel Corporation 8th Gen Core Processor Host Bridge/DRAM Registers (rev 07)
00:01.0 PCI bridge: Intel Corporation 6th-10th Gen Core Processor PCIe Controller (x16) (rev 07)
00:14.0 USB controller: Intel Corporation Cannon Lake PCH USB 3.1 xHCI Host Controller (rev 10)
00:14.2 RAM memory: Intel Corporation Cannon Lake PCH Shared SRAM (rev 10)
00:16.0 Communication controller: Intel Corporation Cannon Lake PCH HECI Controller (rev 10)
00:17.0 SATA controller: Intel Corporation Cannon Lake PCH SATA AHCI Controller (rev 10)
00:1b.0 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #19 (rev f0)
00:1b.4 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #21 (rev f0)
00:1c.0 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #3 (rev f0)
00:1c.3 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #4 (rev f0)
00:1d.0 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #9 (rev f0)
00:1f.0 ISA bridge: Intel Corporation H370 Chipset LPC/eSPI Controller (rev 10)
00:1f.4 SMBus: Intel Corporation Cannon Lake PCH SMBus Controller (rev 10)
00:1f.5 Serial bus controller: Intel Corporation Cannon Lake PCH SPI Controller (rev 10)
00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (7) I219-V (rev 10)
01:00.0 VGA compatible controller: NVIDIA Corporation GP104 [GeForce GTX 1070] (rev a1)
01:00.1 Audio device: NVIDIA Corporation GP104 High Definition Audio Controller (rev a1)
03:00.0 Non-Volatile memory controller: Micron Technology Inc 2550 NVMe SSD (DRAM-less) (rev 01)
05:00.0 PCI bridge: Tundra Semiconductor Corp. Tsi381 PCIe to PCI Bridge (rev 02)
06:00.0 Audio device: Creative Labs Device 0010 (rev 01)
`

I have tested around with "PavuControl" Digital Output IEC958 Output set to SPDIF (no sound and no Digital Data from Soundcard) and Analog Output Stereo IEC958 -> Output set to Speaker (no sound)

The ACM Audio Control Module of the Soundblaster AE-9 wouldn´t be initialized its dead also.

perexg, 2024-12-02:
I'm afraid, but this is issue tracker for alsa-lib (ALSA library). This seems like a driver issue. Unfortunately, because we lack proper documentation for Sound Blaster cards, the driver extensions are developed by volunteers based on the reverse engineering without any active support from the hardware vendor.

Driver history: https://github.com/torvalds/linux/commits/master/sound/pci/hda/patch_ca0132.c

The audio drivers extensions can be discussed in `linux-sound@vger.kernel.org` mailing list.

closed 2024-12-02 by perexg