alsa-project/alsa-lib #425: alsa-lib-1.2.13 prevents chromium-131.0.6778.85 from playing audio on gentoo.

haoyi-codes, 2024-12-02:
I have recently compiled chromium against alsa-lib-1.2.13 for two different systems, however I am unable to play audio. The terminal output shows:

System 1:

```
Failed to create secure directory (/run/user/1011//pulse): Permission denied
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/pcm/pcm_dmix.c:1000:(snd_pcm_dmix_open) unable to open slave
[6429:6429:1130/194540.301739:ERROR:alsa_util.cc(204)] PcmOpen: default,No such file or directory
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/pcm/pcm_dmix.c:1000:(snd_pcm_dmix_open) unable to open slave
[6429:6429:1130/194540.302803:ERROR:alsa_util.cc(204)] PcmOpen: plug:default,No such file or directory
libpng warning: iCCP: known incorrect sRGB profile
Warning: disabling flag --expose_wasm due to conflicting flags
Warning: disabling flag --expose_wasm due to conflicting flags
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/pcm/pcm_dmix.c:1000:(snd_pcm_dmix_open) unable to open slave
[6429:6429:1130/194614.580896:ERROR:alsa_util.cc(204)] PcmOpen: default,No such file or directory
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/pcm/pcm_dmix.c:1000:(snd_pcm_dmix_open) unable to open slave
[6429:6429:1130/194614.582080:ERROR:alsa_util.cc(204)] PcmOpen: plug:default,No such file or directory
libpng warning: iCCP: known incorrect sRGB profile
```

System 2:

```
socket(): Operation not permitted
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission denied)
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4650:(snd_config_update_r) hooks failed, removing configuration
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission denied)
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4650:(snd_config_update_r) hooks failed, removing configuration
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission denied)
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4650:(snd_config_update_r) hooks failed, removing configuration
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission denied)
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4650:(snd_config_update_r) hooks failed, removing configuration
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission denied)
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4650:(snd_config_update_r) hooks failed, removing configuration
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission denied)
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4650:(snd_config_update_r) hooks failed, removing configuration
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission denied)
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4650:(snd_config_update_r) hooks failed, removing configuration
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission denied)
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4650:(snd_config_update_r) hooks failed, removing configuration
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission denied)
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4650:(snd_config_update_r) hooks failed, removing configuration
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission denied)
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4650:(snd_config_update_r) hooks failed, removing configuration
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission denied)
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4650:(snd_config_update_r) hooks failed, removing configuration
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission denied)
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4650:(snd_config_update_r) hooks failed, removing configuration
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission denied)
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4650:(snd_config_update_r) hooks failed, removing configuration
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission denied)
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4650:(snd_config_update_r) hooks failed, removing configuration
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission denied)
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4650:(snd_config_update_r) hooks failed, removing configuration
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission denied)
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4650:(snd_config_update_r) hooks failed, removing configuration
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission denied)
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4650:(snd_config_update_r) hooks failed, removing configuration
[2516:2516:1129/230412.353283:ERROR:alsa_util.cc(204)] PcmOpen: default,No such file or directory
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission denied)
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4650:(snd_config_update_r) hooks failed, removing configuration
[2516:2516:1129/230412.353697:ERROR:alsa_util.cc(204)] PcmOpen: plug:default,No such file or directory
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission den [... cut ...]
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4029:(snd_config_hooks_call) Cannot open shared library (null) (Error loading shared library /lib/libasound.so.2: Permission denied)
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/conf.c:4650:(snd_config_update_r) hooks failed, removing configuration
[2516:2516:1129/230432.095901:ERROR:alsa_util.cc(204)] PcmOpen: plug:default,No such file or directory
libpng warning: iCCP: known incorrect sRGB profile
```

After downgrading alsa-lib to 1.2.12 and recompiling chromium I was able to play audio just fine.

Further information can also be found at: https://bugs.gentoo.org/945749

haoyi-codes, 2024-12-03:
The output for the first command `aplay -D default /dev/zero` is:

```
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/pcm/pcm_dmix.c:1000:(snd_pcm_dmix_open) unable to open slave
aplay: main:850: audio open error: No such file or directory
```

The output for the second command is:

```
aplay: main:850: audio open error: No such file or directory
```

perexg, 2024-12-03:
Not very helpful. Please, show output from `aplay -l` and `cat /proc/asound/cards` and `ls -la /dev/snd/*` .

haoyi-codes, 2024-12-03:
`aplay -l`

```
**** List of PLAYBACK Hardware Devices ****
card 0: Audio [SA9123 USB Audio], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 3: Generic Digital [Generic Digital]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

`cat /proc/asound/cards`

```
 0 [Audio          ]: USB-Audio - SA9123 USB Audio
                      SYC SA9123 USB Audio at usb-0000:00:14.0-3.2, full speed
 1 [HDMI           ]: HDA-Intel - HDA ATI HDMI
                      HDA ATI HDMI at 0x81220000 irq 165
 2 [Edition        ]: USB-Audio - Logi 4K Stream Edition
                      Logi 4K Stream Edition at usb-0000:00:14.0-3.3, super speed
```

`ls -la /dev/snd/*`

```
crw-rw----+ 1 root audio 116,  0 Dec  3 12:03 /dev/snd/controlC0
crw-rw----+ 1 root audio 116, 32 Dec  3 10:36 /dev/snd/controlC1
crw-rw----+ 1 root audio 116, 64 Dec  3 12:03 /dev/snd/controlC2
crw-rw----+ 1 root audio 116, 16 Dec  3 15:46 /dev/snd/pcmC0D0p
crw-rw----+ 1 root audio 116, 51 Dec  3 15:05 /dev/snd/pcmC1D3p
crw-rw----+ 1 root audio 116, 88 Dec  3 15:05 /dev/snd/pcmC2D0c
crw-rw----+ 1 root audio 116,  1 Dec  3 10:36 /dev/snd/seq
crw-rw----+ 1 root audio 116, 33 Dec  3 10:36 /dev/snd/timer

/dev/snd/by-id:
total 0
drwxr-xr-x 2 root root  80 Dec  3 12:03 .
drwxr-xr-x 4 root root 240 Dec  3 12:03 ..
lrwxrwxrwx 1 root root  12 Dec  3 12:03 usb-046d_Logi_4K_Stream_Edition_51C07EAF-03 -> ../controlC2
lrwxrwxrwx 1 root root  12 Dec  3 12:03 usb-SYC_SA9123_USB_Audio-01 -> ../controlC0

/dev/snd/by-path:
total 0
drwxr-xr-x 2 root root 140 Dec  3 12:03 .
drwxr-xr-x 4 root root 240 Dec  3 12:03 ..
lrwxrwxrwx 1 root root  12 Dec  3 12:03 pci-0000:00:14.0-usb-0:3.2:1.1 -> ../controlC0
lrwxrwxrwx 1 root root  12 Dec  3 12:03 pci-0000:00:14.0-usb-0:3.3:1.3 -> ../controlC2
lrwxrwxrwx 1 root root  12 Dec  3 12:03 pci-0000:00:14.0-usbv2-0:3.2:1.1 -> ../controlC0
lrwxrwxrwx 1 root root  12 Dec  3 12:03 pci-0000:00:14.0-usbv3-0:3.3:1.3 -> ../controlC2
lrwxrwxrwx 1 root root  12 Dec  3 10:36 pci-0000:05:00.1 -> ../controlC1
```

It seems the previous commands are working (not sure why they were not earlier):

`aplay -D default /dev/zero`

```
Playing raw data '/dev/zero' : Unsigned 8 bit, Rate 8000 Hz, Mono
^CAborted by signal Interrupt...
aplay: pcm_write:2178: write error: Interrupted system call
```

`aplay -D plughw:0 /dev/zero`

```
Playing raw data '/dev/zero' : Unsigned 8 bit, Rate 8000 Hz, Mono
^CAborted by signal Interrupt...
aplay: pcm_write:2178: write error: Interrupted system call
```

It seems this might be something related with chromium. I will do some more testing and will reopen if need be. Thanks for the help so far and feel free to reopen if any further issues come up.

closed 2024-12-03 by haoyi-codes