alsa-project/alsa-lib #426: aplay doesn't play audio sometimes "unable to open slave"

haoyi-codes, 2024-12-04:
I have initially thought it was a bug that was related to chromium as stated in #425. However, now I realize it has to be something else that is wrong.

`aplay -D default /dev/zero`

```
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/pcm/pcm_dmix.c:1000:(snd_pcm_dmix_open) unable to open slave
aplay: main:850: audio open error: No such file or directory
```

`aplay -D plughw:0 /dev/zero`

```
aplay: main:850: audio open error: No such file or directory
```

`aplay -l`

```
**** List of PLAYBACK Hardware Devices ****
card 0: NVidia [HDA NVidia], device 3: Generic Digital [Generic Digital]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 3: Generic Digital [Generic Digital]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Audio [SA9123 USB Audio], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

`cat /proc/asound/cards`

```
 0 [NVidia         ]: HDA-Intel - HDA NVidia
                      HDA NVidia at 0x81080000 irq 17
 1 [HDMI           ]: HDA-Intel - HDA ATI HDMI
                      HDA ATI HDMI at 0x81220000 irq 165
 2 [Audio          ]: USB-Audio - SA9123 USB Audio
                      SYC SA9123 USB Audio at usb-0000:00:14.0-3.2, full speed
 3 [Edition        ]: USB-Audio - Logi 4K Stream Edition
                      Logi 4K Stream Edition at usb-0000:00:14.0-3.3, super speed
```

`ls -la /dev/snd/*`

```
crw-rw----+ 1 root audio 116,   0 Dec  4 23:18 /dev/snd/controlC0
crw-rw----+ 1 root audio 116,  32 Dec  4 23:18 /dev/snd/controlC1
crw-rw----+ 1 root audio 116,  64 Dec  4 23:18 /dev/snd/controlC2
crw-rw----+ 1 root audio 116,  96 Dec  4 23:18 /dev/snd/controlC3
crw-rw----+ 1 root audio 116,  19 Dec  4 23:18 /dev/snd/pcmC0D3p
crw-rw----+ 1 root audio 116,  51 Dec  4 23:18 /dev/snd/pcmC1D3p
crw-rw----+ 1 root audio 116,  80 Dec  4 23:18 /dev/snd/pcmC2D0p
crw-rw----+ 1 root audio 116, 120 Dec  4 23:18 /dev/snd/pcmC3D0c
crw-rw----+ 1 root audio 116,   1 Dec  4 23:18 /dev/snd/seq
crw-rw----+ 1 root audio 116,  33 Dec  4 23:18 /dev/snd/timer

/dev/snd/by-id:
total 0
drwxr-xr-x 2 root root  80 Dec  4 23:18 .
drwxr-xr-x 4 root root 280 Dec  4 23:17 ..
lrwxrwxrwx 1 root root  12 Dec  4 23:18 usb-046d_Logi_4K_Stream_Edition_51C07EAF-03 -> ../controlC3
lrwxrwxrwx 1 root root  12 Dec  4 23:18 usb-SYC_SA9123_USB_Audio-01 -> ../controlC2                                

/dev/snd/by-path:
total 0
drwxr-xr-x 2 root root 160 Dec  4 23:18 .
drwxr-xr-x 4 root root 280 Dec  4 23:17 ..
lrwxrwxrwx 1 root root  12 Dec  4 23:18 pci-0000:00:14.0-usb-0:3.2:1.1 -> ../controlC2                             
lrwxrwxrwx 1 root root  12 Dec  4 23:18 pci-0000:00:14.0-usb-0:3.3:1.3 -> ../controlC3                             
lrwxrwxrwx 1 root root  12 Dec  4 23:18 pci-0000:00:14.0-usbv2-0:3.2:1.1 -> ../controlC2                           
lrwxrwxrwx 1 root root  12 Dec  4 23:18 pci-0000:00:14.0-usbv3-0:3.3:1.3 -> ../controlC3                           
lrwxrwxrwx 1 root root  12 Dec  4 23:18 pci-0000:01:00.1 -> ../controlC0
lrwxrwxrwx 1 root root  12 Dec  4 23:18 pci-0000:05:00.1 -> ../controlC1
```

For reference I am still able to play audio on my system as pulseaudio is running but for applications that use alsa directly it does not work.

```
Server String: /run/user/1011/pulse/native
Library Protocol Version: 35
Server Protocol Version: 35
Is Local: yes
Client Index: 59
Tile Size: 65472
User Name: user
Host Name: system
Server Name: PulseAudio (on PipeWire 1.2.7)
Server Version: 15.0.0
Default Sample Specification: float32le 2ch 48000Hz
Default Channel Map: front-left,front-right
Default Sink: alsa_output.usb-SYC_SA9123_USB_Audio-01.iec958-stereo
Default Source: alsa_output.usb-SYC_SA9123_USB_Audio-01.iec958-stereo.monitor
Cookie: XXXX:XXXX
```

tiwai, 2024-12-05:
Could you try git bisect of alsa-lib?  Most of changes from 1.2.12 are about MIDI and sequencer stuff, hence they are very likely irrelevant. And the rest is about async handler fix, which doesn't look related, either.

Your stack trace suggests that it might be some memory corruption, so it's not trivial to guess only from that.

haoyi-codes, 2024-12-05:
Yeah I have tried older versions of alsa-lib such as 1.2.12 and even 1.2.11 but they exhibit similar behavior for some reason that is why I had to close my previous issue as it seems something else might be effecting this. Could it be possible that there might be something unrelated to alsa-lib?

tiwai, 2024-12-05:
Right, it implies that the problem isn't in alsa-lib changes but something else, if any.

And the reason why dmix and plughw:0 don't work is that the first card is HDMI audio and it doesn't define the PCM index 0 due to historical reasons. Both default (non-pulse / pipewire) and plughw definitions assume the default access to the PCM index 0 unless specified.

You have no other analog onboard audio but only USB audio? If so, you may reserve the card slot 0 for usb-audio, for example, by passing slots=snd_usb_audio to snd module.

haoyi-codes, 2024-12-05:
Yeah I don't use the HDMI audio or anything like that just the USB audio device. Through pulsemixer I have disabled those devices so audio doesn't play through them but I think that only does it for pulseaudio but not alsa or pipewire.

I will try setting the usb audio device to slot 0 and follow up with you. Are there any other ways to change the slot definitions without probing the module or passing kernel command line parameters?

haoyi-codes, 2024-12-05:
I have removed support for CONFIG_SND_PCI so the output for `aplay -l` would be:

```
**** List of PLAYBACK Hardware Devices ****
card 0: Audio [SA9123 USB Audio], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

However I also use a kvm switch to pass this audio device to a virtual machine whenever I am using it and upon passing the USB audio device back `aplay -l` shows:

```
**** List of PLAYBACK Hardware Devices ****
card 2: Audio [SA9123 USB Audio], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

This causes `aplay -D default /dev/zero` to no longer work:

```
ALSA lib /var/tmp/portage/media-libs/alsa-lib-1.2.13/work/alsa-lib-1.2.13/src/pcm/pcm_dmix.c:1000:(snd_pcm_dmix_open) unable to open slave
aplay: main:850: audio open error: No such file or directory
```

Is there anyway to fix this?

tiwai, 2024-12-05:
Does the first card has a playback stream?  Otherwise it's logical that open for playback fails.

haoyi-codes, 2024-12-05:
> The module options are the basic way to manage the card indices manually.
> 
> Alternatively, you may specify different card index (or ID) for the default PCM card in ~/.asoundrc or whatever alsa-lib configuration, too.

Maybe I would be able to add this to /etc/asound.conf

```
pcm.!default {
    type hw
    card 0
}

ctl.!default {
    type hw
    card 0
}
```

But since card 0 *changes* I think I have to specify what the exact device that I want. And yes I also believe that is why it fails.

closed 2024-12-05 by haoyi-codes

perexg, 2024-12-05:
This is information for reference:

By default, the sound card detection order is not defined. If you like to refer a specific ALSA sound card directly using any native ALSA applications (and you don't want to use a sound server like pulseaudio/pipewire), you can use ASCII card identifier instead card number when you specificy ALSA device.

Card identifiers - use `aplay -l` and the strnig between `card *: ` and ' [' like `KM1U` in the example bellow:

```
card 1: KM1U [KURZWEIL Pro Microphone KM-1U], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```

Then use:

```
pcm.!default {
    type hw
    card KM1U
}

ctl.!default {
    type hw
    card KM1U
}
```

If you need to alter this ASCII identifier (alias), use udev rules and set id attribute. See `/sys/class/sound/card*` subdirectories. Even 'echo "Own_ID" > /sys/class./sound/card0/id` command should work (as root).

Note that serious distributions redirects all I/O to sound severs by default now. So users don't need to do the above step manually.