alsa-project/alsa-lib #429: API memory allocator

sylware, 2024-12-20:
If I recall properly some data returned from the alsa lib API must be free using the C runtime free().

Would it be appropriate to move away from the C allocator to an API install-able memory allocator (like vulkan3D)?

Backward compatibility would be achieved since the "default memory allocator" would be the C one, if the C runtime is around.

Unless that part of the API is already deprecated (and I should not have used it).

perexg, 2024-12-20:
What's the target purpose?

sylware, 2024-12-21:
Huh? For the API To be cleanly independent of the C runtime, like the vulkan3D API.

perexg, 2024-12-21:
If you don't have a real requirement (a code which will benefit from this change), it's really low priority task. You can propose your implementation for the review.

sylware, 2024-12-22:
What? This is not the goal of this issue.

It is an issue to bring this subject on the table, in the mind of alsa people. Because, I realized the alsa user code I wrote require the usage of free() right while I was coding some vulkan3D stuff (which API is very carefull about just that).