alsa-project/alsa-lib #431: alsa-lib-1.2.13 is breaking dssi-1.1.1 because of missing header

kobboi, 2024-12-30:
What was working with 1.2.12 no longer isn't:

```
/usr/include/alsa/ump_msg.h:1119:65: error: unknown type name ‘size_t’
 1119 | int snd_ump_msg_sysex_expand(const uint32_t *ump, uint8_t *buf, size_t maxlen,
      |                                                                 ^~~~~~
/usr/include/alsa/ump_msg.h:13:1: note: ‘size_t’ is defined in header ‘<stddef.h>’; did you forget to ‘#include <stddef.h>’?
   12 | #include <stdint.h>
  +++ |+#include <stddef.h>
   13 |
/usr/include/alsa/ump_msg.h:1120:30: error: unknown type name ‘size_t’
 1120 |                              size_t *filled);
      |                              ^~~~~~
/usr/include/alsa/ump_msg.h:1120:30: note: ‘size_t’ is defined in header ‘<stddef.h>’; did you forget to ‘#include <stddef.h>’?
```

Can this be addressed? Happy holidays!

hhoffstaette, 2025-02-01:
Confirmed via https://bugs.gentoo.org/949127

perexg, 2025-02-01:
It was already fixed in 07cee0ba05179a56764c35975d5822420d4f31f9 .

closed 2025-02-01 by perexg

hhoffstaette, 2025-02-01:
This isn't fixed at all. The header uses a type that is not defined (size_t), and the correct fix is to include stddef.h in ump_msg.h, exactly as the compiler says.

perexg, 2025-02-02:
Okay, I was wrong. The better fix is in https://github.com/alsa-project/alsa-lib/commit/ea8972c83b020d92e1a9f0a5c12eaee159bf6c63 . Those files were never supposed to be included separately.