alsa-project/alsa-lib #439: FiiO K7 USB DAC no Inputs

Starfoxfs, 2025-02-21:
I have problems with Fiio K7 USB DAC in Linux Mint Cinnamon 22.1 (6.8.0-56-generic)

Alsa Version: 1.0.25+dfsg-0ubuntu7
Pipewire: 1.0.5-1ubuntu2
Wireplumber: 0.4.17-1ubuntu4

The USB Dac has this Ports:
2x Line IN (Cinch)
2x Line OUT or Pre OUT(Cinch)
1x Digital IN TOSLINK
1x COAX/RCA IN (Yellow Cinch)

The Problem is:
Alsa didnt find any of this IN Ports. 

There are 4 OUTPUTS:
Analog Stereo works fine with Alsa
Digital IEC958 (thats wrong Device has no Digital Output)
Digital Surround 51 (thats wrong Device has no Digital Output)
Pro Audio works fine with Alsa

Would be nice to have the Inputs.

Heres my Alsa-Info Output:

[alsa-info.txt](https://github.com/user-attachments/files/18907991/alsa-info.txt)

perexg, 2025-02-21:
DAC means Digital Analog Converter. I don't see anything about ADC (Analog Digital Converter) in specs and the device<->host system direction is not advertised in USB descriptors:

<details>
<summary>USB Stream information</summary>

GuangZhou FiiO Electronics Co.,Ltd FiiO K7 at usb-0000:00:14.0-1, high speed : USB Audio

```
Playback:
  Status: Stop
  Interface 1
    Altset 1
    Format: S32_LE
    Channels: 2
    Endpoint: 0x01 (1 OUT) (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
    Bits: 32
    Channel map: FL FR
    Sync Endpoint: 0x81 (1 IN)
    Sync EP Interface: 1
    Sync EP Altset: 1
    Implicit Feedback Mode: No
  Interface 1
    Altset 2
    Format: S16_LE
    Channels: 2
    Endpoint: 0x01 (1 OUT) (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
    Bits: 16
    Channel map: FL FR
    Sync Endpoint: 0x81 (1 IN)
    Sync EP Interface: 1
    Sync EP Altset: 2
    Implicit Feedback Mode: No
  Interface 1
    Altset 3
    Format: SPECIAL DSD_U32_BE
    Channels: 2
    Endpoint: 0x01 (1 OUT) (ASYNC)
    Rates: 44100, 48000, 88200, 96000, 176400, 192000, 352800, 384000
    Data packet interval: 125 us
    Bits: 32
    DSD raw: DOP=0, bitrev=0
    Channel map: FL FR
    Sync Endpoint: 0x81 (1 IN)
    Sync EP Interface: 1
    Sync EP Altset: 3
    Implicit Feedback Mode: No
```

</summary>

Starfoxfs, 2025-02-21:
If i switch to a Analog Input or Optical Input the Device is lost.

Starfoxfs, 2025-02-21:
I tested it in Windows and if you change the input source the K7 USB port switches off. This means that the input sources Coax and Toslink only loop through to the headphones and not as Input Source to the System.

closed 2025-02-21 by Starfoxfs