alsa-project/alsa-lib #441: unclear example how to use remap plugin

dibalavs, 2025-03-01:
Hello. 

I combiled 3 sound cards into one vrurtual 5.1 card for PCM, and now trying to make volume control also to use single virtual device, to set up volume for all chanel of virtual 5.1 card.

 
I am trying to use remap control plugin mentioned at [source](https://github.com/alsa-project/alsa-lib/blob/master/src/control/control_remap.c#L1230)

But I unable to understand what does it mean:

```
ctl.name {
	type remap              # Remap controls
	child STR               # Child name
	# or
	child {                 # Child definition
		type STR
		...
	}
	remap {
		# the ID strings are parsed in the amixer style like 'name="Headphone Playback Switch",index=2'
		SRC_ID1_STR DST_ID1_STR
		SRC_ID2_STR DST_ID2_STR
		...
	}
	map {
		# join two stereo controls to one
		CREATE_ID1_STR {
			SRC_ID1_STR {
				vindex.0 0	# source channel 0 to merged channel 0
				vindex.1 1
			}
			SRC_ID2_STR {
				vindex.2 0
				vindex.3 1	# source channel 1 to merged channel 3
			}
		}
		# split stereo to mono
		CREATE_ID2_STR {
			SRC_ID3_STR {
				vindex.0 0	# stereo to mono (first channel)
			}
		}
		CREATE_ID3_STR {
			SRC_ID4_STR {
				vindex.0 1	# stereo to mono (second channel)
			}
		}
	}
}
```

I do not understand what should I write as SRC and DST?:

```
# the ID strings are parsed in the amixer style like 'name="Headphone Playback Switch",index=2'
SRC_ID1_STR DST_ID1_STR
SRC_ID2_STR DST_ID2_STR
```

Can you give real example without placeholders like: "name", "SRC", "DST", "STR" ?

perexg, 2025-03-01:
> Can you give real example without placeholders like: "name", "SRC", "DST", "STR" ?

It's already in your comment: `'name="Headphone Playback Switch",index=2'`

You can list current mixer IDs using command `amixer -c 0 controls` (don't use numid part)

dibalavs, 2025-03-01:
My example for front channels:

```

pcm.multi {
        type multi;
        slaves.a.pcm "hw:0,0";
        slaves.a.channels 2;
        slaves.b.pcm "hw:3,0";
        slaves.b.channels 2;
        slaves.c.pcm "hw:4,0";
        slaves.c.channels 2;
        bindings.0.slave a;
        bindings.0.channel 0;
        bindings.1.slave a;
        bindings.1.channel 1;
        bindings.2.slave b;
        bindings.2.channel 0;
        bindings.3.slave b;
        bindings.3.channel 1;
        bindings.4.slave c;
        bindings.4.channel 0;
        bindings.5.slave c;
        bindings.5.channel 1;
}

ctl.multi {
	type remap              # Remap controls
	child {                 # Child definition
		type plug
		slave.pcm "multi"
	}
	remap {
		# the ID strings are parsed in the amixer style like 'name="Headphone Playback Switch",index=2'
		'numid=4,iface=MIXER,name="PCM Playback Volume"' 'name="Left front channel volume",index=0'
		'numid=4,iface=MIXER,name="PCM Playback Volume"' 'name="Right fron channel volume",index=1'
	}
	map {
		'name="Left front channel volume",index=0' {
			'numid=4,iface=MIXER,name="PCM Playback Volume"' {
				vindex.0 0	# stereo to mono (first channel)
			}
		}
		'name="Right front channel volume",index=0' {
			'numid=4,iface=MIXER,name="PCM Playback Volume"' {
				vindex.0 1	# stereo to mono (second channel)
			}
		}
	}
}

# This creates a 4 channel interleaved pcm stream based on
# the multi device. JACK will work with this one.

pcm.ttable {
        type route;
        slave.pcm "multi";
        slave.channels 6;
        ttable.0.0 1;
        ttable.1.1 1;
        ttable.2.2 1;
        ttable.3.3 1;
        ttable.4.4 1;
        ttable.5.5 1;
}
# see above.
ctl.ttable {
        type hw;
        card 3;
}

ctl.!default ctl.multi
```

It does not work, alsamixer even not opens:

```
alsamixer -D multi
ALSA lib dlmisc.c:337:(snd_dlobj_cache_get0) Cannot open shared library libasound_module_ctl_plug.so (/usr/lib/aarch64-linux-gnu/alsa-lib/libasound_module_ctl_plug.so: cannot open shared object file: No such file or directory)
cannot open mixer: No such device or address
```

perexg, 2025-03-01:
```
	child {                 # Child definition
		type plug
		slave.pcm "multi"
	}
```

Should be (just use the card you want to use, no pcm is involved):

```
     child "hw:0"
# or
     child { type hw card 0 }
```

The ID names has some conventions, just don't mix different upper/lower case than you see in the amixer dump.

perexg, 2025-03-01:
And combining multiple cards with different clock is utterly bad idea.

perexg, 2025-03-01:
Also use `remap` or `map` for the same element only. In your case, use just `map` section.

dibalavs, 2025-03-01:
Thanks. I am able to launch alsamixer.

```
ctl.multi {
	type remap              # Remap controls
	child "hw:0"
	child "hw:3"
	child "hw:4"
	#remap {
	#	'numid=4,iface=MIXER,name="PCM Playback Volume"' 'name="Left front"'
	#	'numid=4,iface=MIXER,name="PCM Playback Volume"' 'name="Right front"'
	#}
	map {
		'name="Left front"' {
			'numid=4,iface=MIXER,name="PCM Playback Volume"' {
				vindex.0 0	# stereo to mono (first channel)
			}
		}
		'name="Right front"' {
			'numid=4,iface=MIXER,name="PCM Playback Volume"' {
				vindex.0 1	# stereo to mono (second channel)
			}
		}
	}
}
```

I can see new controls for left/right channels. But also exists common combiled PCM control. Is it possible to hide it?

Also as I see,  controls are applied only for device "hw:4" How can I add also spcicify device name in map rules?

> And combining multiple cards with different clock is utterly bad idea.

I understand. But I need this for short voice audio notifications. I think it is ok to small out of sync.

perexg, 2025-03-01:
```
        child "hw:0"
	child "hw:3"
	child "hw:4"
```

This is not allowed and only last config is valid (overwrites the previous one). Only one child can be used, thus you cannot combine and address controls from multiple cards. It's in-place addition to controls per one card. You can probably create a script/daemon to sync depending controls.

dibalavs, 2025-03-01:
Thanks for explanation. Issue can be closed.

closed 2025-03-01 by dibalavs