alsa-project/alsa-lib #447: Difference in behaviour between physical devices and virtual endpoints

reuk, 2025-03-31:
As a disclaimer, there's a good chance this is user error/misunderstanding rather than a bug, but I'd appreciate any insights to help me understand the behaviour I'm seeing.

# Environment

- Fedora 41 x86_64
- alsa-lib-devel-1.2.13-3

# Steps

- Run `aseqdump -u 1` in a terminal
- Connect ProtoZOA hardware via USB
- Launch QjackCtl -> Graph, then connect the ProtoZOA's MIDI 2.0 output port (0) to `aseqdump`'s input port.

[image]

- Observe that `aseqdump` reports a new subscription from `24:0 -> 128:0` (the client IDs may vary, the important thing is that the subscription is between to ports with ID `0`).
- Move one of the controllers on the ProtoZOA. Observe that `aseqdump` reports channel voice events on group 0 with source `24:1`.

[image]

- Connect `aseqdump` to port `1:Group 1 (FB1, FB2)`, and interact with the ProtoZOA. Observe that `aseqdump` still reports channel voice events on group 0 with source `24:1`.
- Now, build and run the [seq-ump-example](https://github.com/alsa-project/alsa-lib/blob/master/test/seq-ump-example.c).
- Connect the ProtoZOA to the ACMESynth input, and the ACMESynth output to `aseqdump` as shown. Observe that `aseqdump` does **not** report channel voice events when interacting with the ProtoZOA.

[image]

- Finally, update the connections as shown, so that `aseqdump` is connected to ACMESynth's `1:Group 1 (Filter 1)` output. Observe that `aseqdump` now reports channel voice events with source `128:1` when interacting with the ProtoZOA.

[image]

# TL;DR

When we subscribe to a USB `MIDI 2.0` port, we receive messages from all of the endpoint's ports. Messages are dispatched to the appropriate port based on the message group.

When we subscribe to a virtual UMP endpoint's `MIDI 2.0` port, this appears to *only* subscribe to that single port, and not to the rest of the ports on the endpoint, so that messages addressed to specific groups get lost.

# Questions

- Is this difference in behaviour between hardware and virtual endpoints intentional?
- What is the "correct" way to subscribe to all UMP events from an endpoint? Should I subscribe just to the `MIDI 2.0` endpoint, or to all ports on the endpoint, or something else?

tiwai, 2025-04-01:
Thanks for the report. I believe it's just a bug in snd_seq_create_ump_endpoint() that forgot to set SND_SEQ_PORT_CAP_UMP_ENDPOINT attribute for MIDI 2.0 port.  The fix patch is below.

[0001-ALSA-seq-Add-missing-UMP-EP-cap-bit-at-snd_seq_creat.patch.txt](https://github.com/user-attachments/files/19545756/0001-ALSA-seq-Add-missing-UMP-EP-cap-bit-at-snd_seq_creat.patch.txt)

reuk, 2025-04-01:
That's great, thanks for the speedy response and fix!

closed 2025-04-01 by tiwai