alsa-project/alsa-lib #451: On output devices, `snd_pcm_open()` makes a bad loud tick.

ewtoombs, 2025-04-13:
Minimal reproduction:

```
#include <alsa/asoundlib.h>
#include <stdbool.h>
int main(int argc, const char **argv)
{
    snd_pcm_t *out;
    assert(snd_pcm_open(&out,
                "default",    // the name of the device being opened
                SND_PCM_STREAM_PLAYBACK, // it is an output stream
                false         // iff nonblocking stream
                ) == 0);
}
```
(Compile with `gcc -o test test.c -lasound`.)

Relevant `lspci`:
```
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
```

`uname`:
```
Linux robert 6.13.8-arch1-1 #1 SMP PREEMPT_DYNAMIC Sun, 23 Mar 2025 17:17:30 +0000 x86_64 GNU/Linux
```

alsa-lib version: 1.2.13-1 (archlinux)

I've heard similar artefacts on many different sound cards, though, across many different architectures, always on linux.
One must wait about five minutes in between trials, or the artefact does not occur.

This isn't a minor detail.
The tick is very loud.
It might even damage speakers in a worst case scenario.

perexg, 2025-04-14:
It looks like a kernel driver issue not alsa library issue. You may discuss it on `linux-sound@vger.kernel.org` mailing list.

closed 2025-04-14 by perexg

ewtoombs, 2025-04-16:
> It looks like a kernel driver issue not alsa library issue. You may discuss it on `linux-sound@vger.kernel.org` mailing list.

Well, OK then, perexg.
I guess you would know.
Thanks for the email address.
I just sent the same report there.