alsa-project/alsa-lib #453: Bug: src/topology/data.c: - Uninitialized 'id', Maybe Coding Error

zkh8227, 2025-04-29:
#Description
       During the code review, I detected an issue with an uninitialized variable in the src/topology/data.c file. In the following calls, I couldn't find an initialization operation for id. Is it possible that the intention here was to call parent->id?
#Error Details
     Bug Type: Uninitialized Variable
     File Path: src/topology/data.c
     Line Number: 1905,1908
     Code Snippet：
    1883 	char id[SNDRV_CTL_ELEM_ID_NAME_MAXLEN];
     ........
    1900 	if (tplg_verify_tuples(tplg, pos, tp->array, tp->size) < 0) {
    1901 		if (tuples) {
    1902 			err = tplg_ref_add(elem, SND_TPLG_TYPE_TOKEN, parent->id);
    1903 			if (err < 0)
    1904 				return err;
    1905 			err = tplg_ref_add(elem2, SND_TPLG_TYPE_TUPLE, id);
    1906 			if (err < 0)
    1907 				return err;
    1908 			err = tplg_ref_add(parent, SND_TPLG_TYPE_DATA, id);
    1909 			if (err < 0)
    1910 				return err;
    1911 			tuples = NULL;
    1912 		}

zkh8227, 2025-04-29:
> No description provided.

perexg, 2025-05-06:
False claim - https://github.com/alsa-project/alsa-lib/blob/master/src/mixer/bag.c#L62 .

Please, try to analyze the code more carefully next time.

closed 2025-05-06 by perexg

zkh8227, 2025-05-08:
> False claim - https://github.com/alsa-project/alsa-lib/blob/master/src/mixer/bag.c#L62 .
> 
> Please, try to analyze the code more carefully next time.

But I was talking about bag_del_all, not bag_del.

perexg, 2025-05-08:
`bag_del_all` removes only all items from the bag, so you can reuse it later. The structure is freed in `bag_free`. Anyway, `bag_del_all` function is not used in alsa-lib's code at the moment.

zkh8227, 2025-05-08:
> `bag_del_all` removes only all items from the bag, so you can reuse it later. The structure is freed in `bag_free`. Anyway, `bag_del_all` function is not used in alsa-lib's code at the moment.

First, bag_del_all only removes elements from the linked list without freeing the memory occupied by these elements. What can be reused is only the head of the linked list (that is, the bag itself), and calling bag_free can only free the bag itself. 
Second, since bag_del_all is not used in the current code, it is recommended to remove this function. Otherwise, if it is used in future code updates, it will lead to memory leakage.

perexg, 2025-05-08:
> First, bag_del_all only removes elements from the linked list without freeing the memory occupied by these elements. What can be reused is only the head of the linked list (that is, the bag itself), [...]

No, bag is handling only pointer list and it does not manage dynamic allocation for those pointers - caller must do that (the structure represented with this pointer may have  a complex layout - so just `free` call may not be enough). The implementation is fine - `bag_del_all` is a counterpart to `bag_del`. I don't see any problem there.

zkh8227, 2025-05-09:
> No, bag is handling only pointer list and it does not manage dynamic allocation for those pointers - caller must do that (the structure represented with this pointer may have a complex layout - so [...]

Yes, the "bag" only handles the pointer list and does not manage the dynamic allocation for those pointers. However, judging from the code of "bag_add" (https://github.com/alsa-project/alsa-lib/blob/master/src/mixer/bag.c#L47) and "bag_del" (https://github.com/alsa-project/alsa-lib/blob/master/src/mixer/bag.c#L62), it is also responsible for the "malloc" and "free" operations of the "bag1_t" structures in the linked list. However, "bag_del_all", which corresponds to "bag_del", does not match "bag_del" in this regard. It only deletes the "bag1_t" structures that were allocated by "bag_add" using "malloc" but does not free them.（You can compare the operations on lines 61 - 61 and line 72. ）

perexg, 2025-05-09:
You're right. My head though that `bag_del` is called from `bag_del_all` which is not this case. Fixed in the above commit. Thanks for your argumentation.