alsa-project/alsa-lib #454: No `appl_ptr` update or `.ack` callback call on driver side when using dmix plugin. Works fine using `hw:`

dbrignoli, 2025-05-02:
This issue affects an out of tree `snd-asihpi` driver which uses a different h/w pointer update implementation compared to the in-tree `snd-asihpi` driver. In particular, this version of the driver uses the `.ack` callback to manage h/w pointer update. Driver source is available for download here <https://www.audioscience.com/internet/download/drivers/released/v4/20/47/hpklinux_4.20.47-g51a97a1a2.tar.gz>.

Using the following `.asoundrc` (with or without `slowptr true`):

```
pcm.testPlugDmix {
	type plug
	slave.pcm "dmixer"
}

pcm.dmixer  {
	type dmix
	ipc_key 1024
	slave {
		pcm "hw:1,0"
		channels 2
		format S32_LE
		period_time 0
		period_size 1024
		buffer_size 4096
		rate 44100
	}
	bindings {
		0 0
		1 1
	}
	# slowptr true
}
```

and running aplay with `-D testPlugDmix` the trace looks like this:

```
1)               |  snd_pcm_ioctl [snd_pcm]() {
 1)               |    snd_pcm_common_ioctl [snd_pcm]() {
 1)               |      snd_pcm_sync_ptr [snd_pcm]() {
 1)   0.388 us    |        pcm_lib_apply_appl_ptr [snd_pcm](); /* = 0x0 */
 1)   1.155 us    |      } /* snd_pcm_sync_ptr [snd_pcm] = 0x0 */
 1)   1.697 us    |    } /* snd_pcm_common_ioctl [snd_pcm] = 0x0 */
 1)   2.055 us    |  } /* snd_pcm_ioctl [snd_pcm] = 0x0 */
```

Running aplay with `-D hw:1` instead the trace looks like this:

```
 0)               |  snd_pcm_ioctl [snd_pcm]() {
 0)               |    snd_pcm_common_ioctl [snd_pcm]() {
 0)               |      __snd_pcm_lib_xfer [snd_pcm]() {
 0)   0.333 us    |        snd_pcm_stream_lock_irq [snd_pcm](); /* = 0x0 */
 0)   0.302 us    |        snd_pcm_stream_unlock_irq [snd_pcm](); /* = 0x0 */
 0)               |        interleaved_copy [snd_pcm]() {
 0)               |          do_transfer [snd_pcm]() {
 0)   4.608 us    |            default_write_copy [snd_pcm](); /* = 0x0 */
 0)   5.249 us    |          } /* do_transfer [snd_pcm] = 0x0 */
 0)   8.835 us    |        } /* interleaved_copy [snd_pcm] = 0x0 */
 0)   0.311 us    |        snd_dma_buffer_sync [snd_pcm](); /* = 0x0 */
 0)   0.299 us    |        snd_pcm_stream_lock_irq [snd_pcm](); /* = 0x0 */
 0)               |        pcm_lib_apply_appl_ptr [snd_pcm]() {
 0)               |          snd_card_asihpi_playback_ack [snd_asihpi]() {
 0)               |            /* asihpi_stream_xfer: pcmC1D0p/sub0 : 4096 */
 0)               |            /* asihpi_stream_state: pcmC1D0p/sub0 in .ack: ALSA: aptr 4096, hptr 0, avail 12288; HPI size 16384, host 4096 (4096), dsp 0 (0, 0), fifo 0, avail 12288, xfer 0 */
 0)   1.376 us    |          } /* snd_card_asihpi_playback_ack [snd_asihpi] = 0x0 */
 0)   1.999 us    |        } /* pcm_lib_apply_appl_ptr [snd_pcm] = 0x0 */
 0)   0.364 us    |        snd_pcm_update_state [snd_pcm](); /* = 0x0 */
 0)   0.299 us    |        snd_pcm_stream_unlock_irq [snd_pcm](); /* = 0x0 */
 0) + 15.415 us   |      } /* __snd_pcm_lib_xfer [snd_pcm] = 0x1000 */
 0) + 16.252 us   |    } /* snd_pcm_common_ioctl [snd_pcm] = 0x0 */
 0) + 16.920 us   |  } /* snd_pcm_ioctl [snd_pcm] = 0x0 */
```

With dmix the driver never sees changes to `appl_ptr` and the `.ack` callback is never called. The driver sets the `SNDRV_PCM_INFO_SYNC_APPLPTR` flag and and I have tried adding `SNDRV_PCM_INFO_EXPLICIT_SYNC` but it makes no difference. Using `dsnoop` on the capture side shows the same symptoms.

Is this expected with dmix? Thank you.

perexg, 2025-05-06:
The ack callback is for standard read/write operation. Direct plugins will set the `stop_threshold` to `boundary` requesting the "free-running" mode from the driver. It's expected that the driver maintains regular transfers on it's own (transfer the samples from DMA buffer without appl_ptr updates) when `stop_threshold` is above `buffer_size`.

closed 2025-05-06 by perexg

dbrignoli, 2025-05-06:
Ack, thank you @perexg.

dbrignoli, 2025-05-06:
@perexg , one more question: could you please point me to the docs about "free-running" mode (and other modes that may exist beyond *it* and standard read/write operation)?