alsa-project/alsa-lib #462: New Card: ESI Juli@ eX

Carlisle96, 2025-07-01:
I am not sure where to ask or how to report this:

There was a new card released ESI-Audio Juli@ eX
https://www.esi-audio.com/products/juliaex/ 

I am wondering if this is supported or planned to be supported. 
It is not yet listed here: https://www.alsa-project.org/wiki/Matrix:Vendor-ESI

Carlisle96, 2025-07-24:
Is the support in the pipeline ? Unfortunately I have ordered the card yesterday :)

sagamusix, 2025-07-24:
Since the link you mentioned in your initial post doesn't list any of 
ESI's PCIe cards, only the original PCI models, I suppose that is what 
might be holding back Juli@ eX support. If the Juli@ XTe was already 
supported (which I don't know, as I don't have one), the Juli@ eX would 
probably be working, too.

Carlisle96, 2025-07-24:
I had a Juli@ XTe recently and it worked flawlessly out of the box. ( Fedora 42 )

Carlisle96, 2025-08-01:
My Juli@ eX is being registered by the system.

`lspci
05:00.0 Multimedia audio controller: VIA Technologies Inc. VT1720/24 [Envy24PT/HT] PCI Multi-Channel Audio Controller (rev 01)`

`aplay -l`
doesn't show anything

```
dmesg | grep -i ice1724
[   12.472059] snd_ice1724 0000:05:00.0: enabling device (0000 -> 0001)
[   12.499233] snd_ice1724 0000:05:00.0: No matching model found for ID 0x494c554a
[   12.500665] snd_ice1724 0000:05:00.0: Invalid EEPROM (size = 64)
[   12.500690] snd_ice1724 0000:05:00.0: probe with driver snd_ice1724 failed with error -5
```

perexg, 2025-08-03:
It appears like a kernel driver issue. You may discuss your problem in linux-sound@vger.kernel.org mailing list.

closed 2025-08-03 by perexg