alsa-project/alsa-lib #465: HDMI card is marked "off" on system resume and stays so until reset

j39m, 2025-07-24:
(Continued from [Pipewire issue 4809](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/4809).)

# Basic Symptoms

When resuming, my device often fails to continue playback over my HDMI output. Higher-level utils (`pavucontrol`, `pw-dump`, etc.) report that the corresponding "sink" is not present (until I reset the output).

Pipewire dev `@pvir` read my logs thusly:

<details>

<summary>`alsactl monitor` and `amixer`</summary>

Last state update after resume:

```
2025-07-24 22:15:16,844: alsactl
node hw:0, #7 (0,0,0,HDMI/DP,pcm=7 Jack,0) VALUE

2025-07-24 22:15:16,854: amixer
numid=7,iface=CARD,name='HDMI/DP,pcm=7 Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=off
```

The next update then comes only when workaround is run:

```
2025-07-24 22:16:02,772: alsactl
node hw:0, #12 (3,7,0,ELD,0) VALUE INFO

2025-07-24 22:16:02,781: amixer
numid=7,iface=CARD,name='HDMI/DP,pcm=7 Jack'
  ; type=BOOLEAN,access=r-------,values=1
  : values=on
```

It appears there was no notification for the off->on transition, or the off->on only happens when doing the workaround cycling displays, so probably this is ALSA or HDMI bug.

</details>

I haven't tried waiting longer than this, but I assume that if ALSA thinks the card is still gone ~45s after resume, it's already a bit too far gone to be worried about exact timing.

# System readout

<details>

```
[j39m@flaglock6 ~/Downloads/tmp]
$ rpm -q alsa-lib intel-audio-firmware
alsa-lib-1.2.14-3.fc42.x86_64
intel-audio-firmware-20250708-1.fc42.noarch
[j39m@flaglock6 ~/Downloads/tmp]
$ inxi -A
Audio:
  Device-1: Advanced Micro Devices [AMD/ATI] Raven/Raven2/Fenghuang HDMI/DP
    Audio driver: snd_hda_intel
  Device-2: Advanced Micro Devices [AMD] Audio Coprocessor
    driver: snd_pci_acp3x
  Device-3: Advanced Micro Devices [AMD] Family 17h/19h/1ah HD Audio
    driver: snd_hda_intel
  API: ALSA v: k6.15.6-200.fc42.x86_64 status: kernel-api
  Server-1: PipeWire v: 1.4.6 status: active
```

</details>

perexg, 2025-08-03:
It appears like a kernel driver issue. You may discuss your problem in `linux-sound@vger.kernel.org` mailing list.

closed 2025-08-03 by perexg

superm1, 2025-09-05:
Just to close the loop here - this ended up being an amdgpu regression.  Here's [the fix](https://lore.kernel.org/amd-gfx/CADnq5_POMonCMXc43eJXOuAvJX1B_h-dVG5xpJCGrYKKa++Pkg@mail.gmail.com/T/#m23451ae6b144e5672c66fb39203e5c944012667c).

perexg, 2025-09-05:
Thanks @superm1 .